jimmykane/fit-parser #4: left_right_balance undefined from SRM PC8 fit file

joecorkerton, 2019-06-16:
When parsing any file generated from my PC8 everything seems to work, except the left_right_balance field which returns as undefined. 

Not sure why this would be the case but the data is definitely in there as it can be parsed by Golden Cheetah correctly and it is displayed on the PC8.

[image]

jimmykane, 2019-06-16:
Would you mind giving me the fit file that contains that info ?

joecorkerton, 2019-06-16:
Here's one example

[2019_04_19_11_12_12.fit.zip](https://github.com/jimmykane/fit-parser/files/3294105/2019_04_19_11_12_12.fit.zip)

jimmykane, 2019-06-16:
Great, thanks @joecorkerton  I ll give it a look as soon as I can.

jimmykane, 2019-08-27:
> @jimmykane <https://github.com/jimmykane> I don't see any commits since
> July 26th?

I might have not pushed them. Oups. But the module is published. (i am
using now l/r balance)

jimmykane, 2019-08-27:
When home will see what boo boo i made

joecorkerton, 2019-08-27:
@jimmykane I don't see any commits since July 26th?

jimmykane, 2019-08-27:
This should be fixed now @joecorkerton  can you validate?

jimmykane, 2019-08-27:
Should be fine now @joecorkerton

closed 2019-09-09 by jimmykane