jimmykane/fit-parser #5: Calculation of offset and scale incorrect?

cycore, 2019-06-17:
The FIT sdk documentation describes how to apply scale and offset to fields. 

> the binary quantity is divided by the scale factor and then the offset is **subtracted**, yielding a floating point quantity

compare _D00001275 Flexible & Interoperable Data Transfer (FIT) Protocol Rev 2.3.pdf_, section 4.4

In binary.js, you use `data / scale + offset` (adding offset instead of subtracting). Is this correct? I could not test it, because my fit file doesn't contain any fields that use offsets.

jimmykane, 2019-06-18:
That is correct. Take a look at `fit.js`

We add the offsets as negative numbers: 

Eg 

```
      49: { field: 'avg_altitude', type: 'uint16', scale: 5, offset: -500, units: 'm' },
      50: { field: 'max_altitude', type: 'uint16', scale: 5, offset: -500, units: 'm' },
```

Are you facing some issue?

cycore, 2019-06-18:
Ah I see, so the minus sign is already accounted for.
I did not encounter any issue. I was only wondering if this might be a bug.
Thanks for checking!

closed 2019-06-18 by cycore

jimmykane, 2019-06-18:
Good that you brought it up actually. Could be the case for some future bug :-)