jimmykane/fit-parser #6: Field definitions?

ScottChapman, 2019-07-24:
Is there any place where the fields are explained? Most are pretty self-explanitory, but a bunch are not. For example, while it is obvious the starting location for a session/lap there are two possibilities for the ending location (`swc` or `nec`). No idea what those are...

Are you aware of any documentation anywhere where these fields are explained?

jimmykane, 2019-07-24:
Hey @ScottChapman I think those are documented on Garmins FIT SDK. 
The parser just "parses" those. 

https://www.thisisant.com/resources/fit 

Download the SDK and search for the definitions there. 

I understand that its not optimal but there are too many fields provided by Garmin to keep them documented here. 

What do you think ?

ScottChapman, 2019-07-24:
Thanks, that actually helps. SWC = South West Corner, NEC = North East Corner.

One thing I noticed is that while I have starting coordinated (session and laps) I do not have ending coordinates. I suppose I can make up for that by using the start/end of the records...

closed 2019-07-24 by ScottChapman