jimmykane/fit-parser #8: Feature request

Abasz, 2019-09-03:
Hi, I was wondering whether you are planing to implement encode for fit files from the js object ? I Your parser is the best I could find (I tried a lot :)) but I would like to save the changes back to an actual fit file after editing.

thanks

jimmykane, 2020-07-27:
Atm quite difficult but PR's are welcome. 

That said for export to GPX you could check https://github.com/sports-alliance/sports-lib that leverages this lib :-D

udos, 2020-12-14:
@Abasz, you could handle writing back to the file via the File System Access API. this article gives  a nice introduction: https://web.dev/file-system-access/

Abasz, 2020-12-15:
Thanks for the reply, the problem is not that I am not able to save a file on hard disk, but rather the encoding of a json to FIT format binary.

I have not tried this for a while, but if I remember correctly my issue was that if I decode the FIT file to JSON change something in the json there is no way to encode back to FIT binary.

I actually started using a pyhoton [package](https://opensource.quarq.com/fit_json/) for this purpose. It correctly decodes and then encodes it back.

udos, 2020-12-15:
ok. my misunderstanding. sorry about this.
note: I'm not involved with this project, just saw the issue and thought I could be of help. I was not :|