jimmykane/fit-parser #11: TypeError: Cannot read property 'globalMessageNumber' of undefined

ga262, 2019-10-19:
Hello,
Thanks for this great fit-parser!

I sometimes get the following error for Garmin files (probably from a Garmin 935 watch):
TypeError: Cannot read property 'globalMessageNumber' of undefined
2019-10-19 15:02:     at readRecord (/node_modules/fit-file-parser/dist/binary.js:324:60)
2019-10-19 15:02:     at FitParser.parse (/node_modules/fit-file-parser/dist/fit-parser.js:111:50)

It corresponds to the following line:
var message = (0, _messages.getFitMessage)(messageType.globalMessageNumber);

Any idea on how to solve it?

Best,

jimmykane, 2019-10-20:
Hey let's look at this. Can you share perhaps the fit file with the issue ?

ga262, 2019-11-26:
Thanks for your answer.

Actually it looks like I got an error from Garmin when I was fetching a FIT file from their API, with an unknown UserAccessToken.

The buffer had the following characteristics:
ArrayBuffer { byteLength: 42 }
{"errorMessage":"Unknown UserAccessToken"}

I was sending this buffer directly to the fit-parser, that's why it gave this error: "TypeError: Cannot read property 'globalMessageNumber' of undefined" and "Missing '.FIT' in header".

I think it can be possible to return directly the errorMessage from the fit-parser, but at least for now, I check if the bufferSize is larger enough to be a FIT file.

closed 2019-11-26 by ga262