jimmykane/fit-parser #15: unable to parse downloaded fit file

MichaelLiss, 2020-04-30:
Hi,

I am trying to use your fit file parser and ran into a little issue
I am hoping you will be able to shed some light on.

I am using the garmin health-api and downloading a FIT file directly through the REST api.  Here is an example call 
```javascript
https://healthapi.garmin.com/wellness-api/rest/activityFile?id=86711045091
```

Of course there is a bunch of stuff that needs to occur before you can download the fit file, but rest assured, I have worked with Garmin to get all of the "rest of the stuff" hammered out and I am getting FIT file data from the REST api.

Here is what the fit file data looks like when I view it in my debugger:

[image]

Here is the code where I am using your object to directly parse the downloaded data from the REST call:

```javascript
const internalParse = async (**binary**) => {
    
    // Create a FitParser instance (options argument is optional)
    var fitParser = new FitParser({
        force: true,
        speedUnit: 'km/h',
        lengthUnit: 'km',
        temperatureUnit: 'kelvin',
        elapsedRecordField: true,
        mode: 'cascade',
        });
        
        var error = null;
        var data = null;
        // Parse your file
        try{
            fitParser.parse(**binary**, function (error, data){
```

When I try to read this data using your object I get the following error:
```javascript
Incorrect header size
Missing '.FIT' in header
```

My question to you:
Is there a way that I can save the contents of the REST api call to disk, then read it from disk OR is there a way I can simply pass that data to your object for parsing?

MichaelLiss, 2020-04-30:
Just so you know... I do have code that uses your object and **_successfully_** parses a file that is read from the hard drive - so I know your object works when reading files from hard drive. :-)

```javascript
    var fName = path.join(__dirname, './4861335825.fit');

    fs.readFile(fName, function (err, content) {

        console.log('reading fit file - content',content);
 
        // Create a FitParser instance (options argument is optional)
        var fitParser = new FitParser({
          force: true,
          speedUnit: 'km/h',
          lengthUnit: 'km',
          temperatureUnit: 'kelvin',
          elapsedRecordField: true,
          mode: 'cascade',
        });
        
        // Parse your file
        fitParser.parse(content, function (error, data) {
```

MichaelLiss, 2020-05-01:
If you use Postman with this url  ( note this data will only be good for 7 days ... Garmin policy, not mine )

https://www.myeventtimes.com/api/garmin/testOAuthFitFile

and this raw JSON body:

```Javascript
{
    "_id": "5ea4a671d76a5f00177f6091",
    "dateProcessed": null,
    "userId": "5e6c2dddad72870c84f8476b",
    "userAccessToken": "21b3192e-d051-43b9-9f15-4d7b9937a2d6",
    "summaryId": "54318861003",
    "fileType": "FIT",
    "callbackURL": "https://healthapi.garmin.com/wellness-api/rest/activityFile?id=86711045091",
    "startTimeInSeconds": 1587848764,
    "dateCreated": "2020-04-25T21:06:57.748Z",
    "__v": 0
}
```

[image]

MichaelLiss, 2020-05-01:
I see this in the binary.js file:

// TODO: handle compressed header ((recordHeader & 128) == 128)

MichaelLiss, 2020-05-01:
I thought it might be that utf8 was the wrong encoding... so I tried this:

```javascript
    var encoding = 'utf8';
    encoding = 'utf16le';
    const buf1 = Buffer.from(binary, encoding);
```

But that makes the parser have a wrong fileTypeString:

[image]

jimmykane, 2020-05-01:
Hey. Could you try the option force ? That should allow that block of code
to exacute.

Sorry I am ooo and might take a few to reply properly.

Thanks in advance for reporting those issues.

MichaelLiss, 2020-05-01:
I think I figured it out:

https://stackoverflow.com/questions/61543176/why-is-the-content-of-these-two-buffers-different?noredirect=1#comment108864787_61543176

closed 2020-05-01 by MichaelLiss

jimmykane, 2020-05-01:
I have the exact same code on other lib , that consumers this lib (fit
parser) but also extends for gpx and tcx are you interested ?

MichaelLiss, 2020-05-01:
I need the information that the activity file provides and the FIT file format is the only format that is available for activity file download.
 
Additionally, the fit file format is WAY smaller than a gpx or tcx, 

Lastly, parsing it (much thanks to your parser) is super quick ... so I can just parse on demand when I need to get data

so the short answer is : thanks, but no.