jimmykane/fit-parser #16: forerunner 945 not listed as product

MichaelLiss, 2020-05-01:
Hi again,

I am successfully using the parser now !  (wow... such a learning curve on Java)

Anyway, I downloaded an activity, parsed it and got some results.  However, I think the FR945 is not in the listing of products

I assume 'product' in the following output is going to show fr945 ??

```javascript
"file_id": {
        "serial_number": 3996694396,
        "time_created": "2020-04-30T22:16:48.000Z",
        "manufacturer": "garmin",
        "product": 3113,
        "type": "activity"
    },
```

I see a 'product' like category in fit.js that contains an fr920xt, an edge 280 (both of which I own), the fr935 and fenix5x (both of which I know are watches)... but I don't see the fr945 ? 

```javascript
    garmin_product: {
      0: 'hrm_bike',
      1: 'hrm1',
      <snip>
      1765: 'fr920xt',
      <snip>
      2530: 'edge_820',
     <snip>
      2604: 'fenix5x',
      2606: 'vivo_fit_jr',
      2691: 'fr935',
```

I tried putting this code in and re-running the parser, but it did not pick it up:

```javascript
      2859: 'descent',
      3113: 'fr945',
      10007: 'sdm4',
```

MichaelLiss, 2020-05-01:
to verify that I am not going insane...

I changed this:

```javascript
    manufacturer: {
      0: 0,
      1: 'garmin',
```
to this:

```javascript
    manufacturer: {
      0: 0,
      1: 'garmin_TEST',
```

and got this, so I know I am not too crazy... LOL

```javascript
"file_id": {
        "serial_number": 3996694396,
        "time_created": "2020-04-30T22:16:48.000Z",
        "manufacturer": "garmin_TEST",
        "product": 3113,
        "type": "activity"
    },
```

jimmykane, 2020-05-01:
Yes correct. There are some devices missing. Garmin adds those to each fit
sdk release

Typically I download their sdk then find the unknown device (new added) and
so on. Cumbsersome but it works. There should be some device listing in
this lib.

You just have to change that , and then build (or edit the lib produce)

MichaelLiss, 2020-05-01:
I changed the fit.js file... but it doesn't seem to be picking it up for some reason...

```javascript
      2859: 'descent',
      3113: 'fr945',
      10007: 'sdm4',
```

output:

```javascript
    "file_id": {
        "device_id": 3996694396,
        "time_created": "2020-04-30T22:16:48.000Z",
        "manufacturer": "garmin",
        "product": 3113,
        "type": "activity"
    },
```

MichaelLiss, 2020-05-04:
I just ran another test with a fit file from a Garmin 920xt;

this is what I got:

```javascript
  "file_id": {
    "serial_number": 3896976736,
    "time_created": "2020-05-04T20:17:04.000Z",
    "manufacturer": "garmin",
    "product": 1765,
    "type": "activity"
  }
```

This is what is in the fit.js file
```javascript
      1752: 'hrm_run',
      1765: 'fr920xt',
      1821: 'edge510_asia',
```

jimmykane, 2020-05-05:
Could you provide a fit file ?

MichaelLiss, 2020-05-05:
I can... I have one for a garmin 920xt and a 945.

The 920xt has actuall run data in it.
The 945 I just sat on my bed and clicked start / stop so there is no distance traveled.

here they are

[FitFiles.zip](https://github.com/jimmykane/fit-parser/files/4581931/FitFiles.zip)

MichaelLiss, 2020-05-06:
Did those help ?

jimmykane, 2020-05-06:
Will check it out and let you know

jimmykane, 2020-05-06:
Ok I see the issue here. Actually this was never implemented. 

Meaning that the "matching" happens in the consumer lib. (so far) 

So those ID are not used inside the lib.

Something could be done about that though.

MichaelLiss, 2020-05-11:
Can you update the code so it works with your component here ?

I don't know what 'consumer lib' means or how it ties into this component

Thanks !

MichaelLiss, 2020-05-12:
Hi....

I have been thinking about this.. and for backward compatibility (and stuff that doesn't have a name yet)  what would be 'really' nice is to have both.

```javascript
"file_id": {
        "device_id": 3996694396,
        "time_created": "2020-04-30T22:16:48.000Z",
        "manufacturer": "garmin",
        "product": 3113,
        "productName": "fr945",
        "type": "activity"
    },
```

in the case of no productName: 

```javascript
"file_id": {
        "device_id": 3996694396,
        "time_created": "2020-04-30T22:16:48.000Z",
        "manufacturer": "garmin",
        "product": 3113,
        "productName": "",
        "type": "activity"
    },
```

MichaelLiss, 2020-05-17:
Hello again... 
I was wondering if you have some time to address this ?
I am getting to the point where this issue is starting to block the progress on my project.
(or maybe you could send some instructions on how to modify this so I can update it... and I will try to do the work???)

MichaelLiss, 2020-05-29:
Hi...

I am going to uninstall your module and use Garmin's parser and an AWS micro-service, as it doesn't seem you are going to work on this any time soon.