jimmykane/fit-parser #17: serial number is really device ID

MichaelLiss, 2020-05-01:
Hi again

Just spoke with Garmin:

Elena Kononova (Garmin Health)
May 1, 17:29 CESTOk, 
Seems that we have inconsistent naming, it should be renamed into device id...I see from my internal page that your 'device id' is 3996694396, which is different from serial number

I think to change that the fit.js should be modified here?

```javascript
    3: { field: 'serial_number', type: 'uint32z', scale: null, offset: '', units: '' },
```

to this:
```javascript
    3: { field: 'device_id', type: 'uint32z', scale: null, offset: '', units: '' },
```

I think this is what should be shown:

```javascript
    "file_id": {
        "device_id": 3996694396,
        "time_created": "2020-04-30T22:16:48.000Z",
        "manufacturer": "garmin",
        "product": 3113,
        "type": "activity"
    },
```

jimmykane, 2020-05-06:
If you like you can send a PR :-)

MichaelLiss, 2020-05-17:
Hello again... 
I was wondering if you have some time to address this ?
I am getting to the point where this issue is starting to block the progress on my project.
(or maybe you could send some instructions on how to modify this so I can update it... and I will try to do the work???)

MichaelLiss, 2020-05-29:
Hi...

I am going to uninstall your module and use Garmin's parser and an AWS micro-service, as it doesn't seem you are going to work on this any time soon.

PR #22 (2020-08-18): fix: rename serial_number to device_id
Fixes the issue where Garmin GPS device_id was being reported as
serial_number;

Closes #17
commit 63c7a914 (2020-08-18): fix: rename serial_number to device_id

Closes #17
closed 2020-08-18 by jimmykane