jimmykane/fit-parser #18: cascade vs list : device_infos discrepancy

MichaelLiss, 2020-05-01:
Hi again,

Please forgive my ignorance if this is supposed to behave this way, but I am comparing the output from the 'cascade' option and the 'list' option and noticed a discrepancy between the two.  The 'list' version has multiple device_info records and the cascade does not have any?

[image]

Here is the full output from both:

**Cascade**

```javascript
{
  "file_id": {
    "device_id": 3996694396,
    "time_created": "2020-04-30T22:16:48.000Z",
    "manufacturer": "garmin",
    "product": 3113,
    "type": "activity"
  },
  "file_creator": {
    "software_version": 440
  },
  "device_settings": {
    "utc_offset": 0,
    "time_offset": 4294942096,
    "active_time_zone": 0,
    "time_zone_offset": 0,
    "mounting_side": "left"
  },
  "user_profile": {
    "weight": 74.8,
    "gender": "male",
    "height": 0.00183,
    "language": "english",
    "elev_setting": "statute",
    "weight_setting": "statute",
    "resting_heart_rate": 0,
    "hr_setting": "max",
    "speed_setting": "statute",
    "dist_setting": "statute",
    "position_setting": "degree_minute_second",
    "temperature_setting": "statute",
    "height_setting": "statute"
  },
  "sport": {
    "name": "Run",
    "sport": "running",
    "sub_sport": "generic"
  },
  "zones_target": {
    "functional_threshold_power": 207,
    "max_heart_rate": 180,
    "threshold_heart_rate": 167,
    "pwr_calc_type": "percent_ftp"
  },
  "activity": {
    "timestamp": "2020-04-30T22:17:03.000Z",
    "total_timer_time": 9.199,
    "local_timestamp": "2020-04-30T15:17:03.000Z",
    "num_sessions": 1,
    "type": "manual",
    "event": "activity",
    "event_type": "stop",
    "sessions": [
      {
        "timestamp": "2020-04-30T22:17:03.000Z",
        "start_time": "2020-04-30T22:16:48.000Z",
        "start_position_lat": 37.711845925077796,
        "start_position_long": -121.72617522999644,
        "total_elapsed_time": 9.199,
[... 437 lines cut ...]
    },
    {
      "timestamp": "2020-04-30T22:16:58.000Z",
      "product": 28745,
      "software_version": 1.28,
      "device_index": 4,
      "source_type": "local"
    },
    {
      "timestamp": "2020-04-30T22:16:58.000Z",
      "manufacturer": "garmin",
      "product": 3114,
      "software_version": 4.01,
      "device_index": 5,
      "device_type": "environment_sensor_legacy",
      "source_type": "local"
    }
  ],
  "developer_data_ids": [],
  "field_descriptions": [],
  "hrv": [],
  "dive_gases": [],
  "course_points": []
}
```

jimmykane, 2020-05-01:
Will have to check this

Hold on there. Just need some proper time and as a new dad its hard to find

MichaelLiss, 2020-05-01:
Congratulations !

and I understand !

Mike

MichaelLiss, 2020-05-17:
Hello again... 
I was wondering if you have some time to address this ?
I am getting to the point where this issue is starting to block the progress on my project.
(or maybe you could send some instructions on how to modify this so I can update it... and I will try to do the work???)

jimmykane, 2020-05-20:
I am a bit on vacation but I'll take a look on this one as indeed looks
wierd.

I hope end of weekend.

Don't worry this lib is not forgotten , it's actually used by my main
project, if you want to take a look at it quantified-self.io

Is that ok for you?

MichaelLiss, 2020-05-29:
Hi...

I am going to uninstall your module and use Garmin's parser and an AWS micro-service, as it doesn't seem you are going to work on this any time soon.

jimmykane, 2020-07-26:
Hey ! I am back! Should I go on fixing this ?

MichaelLiss, 2020-07-27:
yes

jimmykane, 2020-07-27:
Should be fixed with 1.7.0

commit 81c8a16c (2020-07-27): fix: sports and cascade mode

fixes #18 and #19
closed 2020-07-27 by jimmykane