jimmykane/fit-parser #19: Why does multi-sport come up as run ?

MichaelLiss, 2020-05-19:
Hi,

I just did a "triathlon" and ran it through the parser, Garmin thinks it is a multi-sport: 

## Garmin Results
[image]

## Fit-file-parser results
However the fit-file-parser thinks the sport name is a "Run" and activity type of 'auto_multi-sport':

```javascript
 \"fileId_type\": \"activity\",
 \"sport_Name\": \"Run\",
 \"sport_Sport\": \"running\",
 \"sport_SubSport\": \"generic\",
 \"activity_Type\": \"auto_multi_sport\"

```

# Other types of activities for comparison
I did a swim, a bike and run for comparison

## ONLY Swim (open water)
here is what I get for just a swim:
```
 \"fileId_type\": \"activity\",
 \"sport_Name\": \"Open Water\",
 \"sport_Sport\": \"swimming\",
 \"sport_SubSport\": \"open_water\",
 \"activity_Type\": \"manual\"
```

## ONLY Bike
here is what I get for just a bike:
```
 \"fileId_type\": \"activity\",
 \"sport_Name\": \"Bike\",
 \"sport_Sport\": \"cycling\",
 \"sport_SubSport\": \"generic\",
 \"activity_Type\": \"manual\"
```

## ONLY Run
here is what I get for just a run:
```
 \"fileId_type\": \"activity\",
 \"sport_Name\": \"Run\",
 \"sport_Sport\": \"running\",
 \"sport_SubSport\": \"generic\",
 \"activity_Type\": \"manual\"
```

## Multi-sport fit file
Here is the FIT file for the multi-sport activity 
[4958669219.zip](https://github.com/jimmykane/fit-parser/files/4653787/4958669219.zip)

MichaelLiss, 2020-05-29:
Hi...

I am going to uninstall your module and use Garmin's parser and an AWS micro-service, as it doesn't seem you are going to work on this any time soon.

jimmykane, 2020-05-29:
Hey no worries. You can always keep the watching on for releases. I will be
working and closing these tickets but for now, during an pandemic and my
son I am enjoying this time.

I am typically a workaholic in a way but unfortunately not this time.

The only reason I reply in such a personal matter here, is to indicate that
typically I commit every day 🙏

jimmykane, 2020-07-26:
Hey should I proceed with this ?

jimmykane, 2020-07-26:
The file you have provided has only one session. Did you export the whole activity or one session here?

MichaelLiss, 2020-07-27:
It is a FIT file for a multi-sport activity

jimmykane, 2020-07-27:
@MichaelLiss 
Looking better (sorry for the confusion) the decoded file has 5 sessions. Each session holds the sport name. 

If an activity has more than 1 session it should be treated by the consuming library as a multisport or multisession etc. 

Here is how the activity of that file looks like 
[image]

jimmykane, 2020-07-27:
Also on the activity object there is `type: "auto_multi_sport"`

jimmykane, 2020-07-27:
Should be now fixed. Look under sports

closed 2020-07-27 by jimmykane