jimmykane/fit-parser #21: How to use this module?

sylvanoMTL, 2020-07-07:
I have tried this fit-file-parser, but it seems to return default fields. 
My FIT activity contains other field that have not been parsed. 

Should I modified the options in the `var fitParser = new FitParser({});`?

Where can I find these options? Should I contact the device manufacturer to know some variable names?

In my case, I am using the Nielsen-Kellerman Speedcoach 2 which retrieve GPS data, Splits/500m, stoke rate and data from the NK empower: Catch angle, slip angle, Force Max, ForceMax Angle, etc...   Approx 13 different metrics.

Kind regards, 
Sylvain

jimmykane, 2020-07-07:
Hey can you link the fit file to check what's going on?

sylvanoMTL, 2020-07-07:
Wow, you are so eactive ! please follow this [link]( https://github.com/sylvanoMTL/node-red-fit-file-parser-test)s

jimmykane, 2020-07-07:
Getting a 404 here

sylvanoMTL, 2020-07-07:
[Updated link](https://github.com/sylvanoMTL/node-red-fit-file-parser-tests)

jimmykane, 2020-07-07:
I can see pretty much: 

```
SUCCESS
# File size: 2396, protocol version: 1.00, profile_version: 13.10
# File header CRC: expected=0x0920, calculated=0x0920
= TYPE=0 NAME=file_id NUMBER=0
--- serial_number=2393792=2393792
--- time_created=960456643=2020-06-07T09:30:43Z
--- product=0=0
--- type=4=activity
--- manufacturer=87=nielsen_kellerman
==
= TYPE=0 NAME=device_info NUMBER=23
--- timestamp=960456643=2020-06-07T09:30:43Z
--- device_index=0=creator
--- manufacturer=87=nielsen_kellerman
--- serial_number=2393792=2393792
--- product=0=0
--- software_version=132=1.32
==
= TYPE=0 NAME=event NUMBER=21
--- timestamp=960456643=2020-06-07T09:30:43Z
--- event=0=timer
--- event_type=0=start
--- event_group=0=0
--- timer_trigger=0=manual
==
= TYPE=0 NAME=record NUMBER=20
--- timestamp=960456644=2020-06-07T09:30:44Z
--- position_lat=613267626=51.4034986 deg
--- position_long=-4073650=-0.3414494 deg
--- speed=850=3.060 km/h
--- distance=215=2.15 m
--- cycle_length=215=2.15 m
--- cadence=37=37 rpm
--- total_cycles=1=1 cycles
--- heart_rate=123=123 bpm
--- power=500=500 watts
==
= TYPE=0 NAME=record NUMBER=20
--- timestamp=960456645=2020-06-07T09:30:45Z
--- position_lat=613267176=51.4034609 deg
--- position_long=-4073091=-0.3414025 deg
--- speed=2410=8.676 km/h
--- distance=745=7.45 m
--- cycle_length=18=0.18 m
--- cadence=47=47 rpm
--- total_cycles=2=2 cycles
--- heart_rate=122=122 bpm
--- power=492=492 watts
==
= TYPE=0 NAME=record NUMBER=20
--- timestamp=960456647=2020-06-07T09:30:47Z
--- position_lat=613266635=51.4034155 deg
--- position_long=-4072410=-0.3413455 deg
--- speed=3910=14.076 km/h
--- distance=1387=13.87 m
--- cycle_length=130=1.30 m
--- cadence=43=43 rpm
--- total_cycles=3=3 cycles
--- heart_rate=124=124 bpm
--- power=487=487 watts
==
= TYPE=0 NAME=record NUMBER=20
--- timestamp=960456648=2020-06-07T09:30:48Z
--- position_lat=613265976=51.4033603 deg
--- position_long=-4071584=-0.3412762 deg
--- speed=4590=16.524 km/h
--- distance=2167=21.67 m
[... 835 lines cut ...]
==
= TYPE=0 NAME=event NUMBER=21
--- timestamp=960456766=2020-06-07T09:32:46Z
--- event=8=session
--- event_type=9=stop_disable_all
--- event_group=1=1
--- data=1=1
==
= TYPE=0 NAME=activity NUMBER=34
--- timestamp=960456766=2020-06-07T09:32:46Z
--- total_timer_time=123000=123.000 s
--- num_sessions=1=1
--- type=0=manual
--- event=26=activity
--- event_type=1=stop
==
# CRC: expected=0xB9DA, calculated=0xB9DA
```

Typically those "extra data" are set in the developer fields from the manifacturer. 
This file doesn't contain any. 

Perhaps the Nielsen-Kellerman Speedcoach 2 doesn't export them? 

I am quite sure that those are not in the FIT file.

sylvanoMTL, 2020-07-07:
I see, I will ask the NK support team to know more.

sylvanoMTL, 2020-07-07:
Confirmation from NK: 

"The only oarlock measurement exported into the FIT file is power.  There needs to be an associated FIT field defined and that was the only one we could associate with the oarlock’s data.  All the other data is in the CSV files."

I am closing the post. 
Thanks for your help

closed 2020-07-07 by sylvanoMTL