jimmykane/fit-parser #27: Incorrect sport mapping: hang gliding instead of paragliding

NainDeathlegs, 2021-04-15:
Importing a paragliding activity from Suunto to qualified-self results in a hang-gliding activity.
Either Suunto exports it with the wrong sport (which is 26) or the mapping is incorrect. 
If so then the issue should be fixed by the following change:
    fit.js:1285 `26: 'hang_gliding',` is supposed to be `26: 'paragliding',`

At least Suunto does not support hang gliding as it is not that popular any more but there is a test case that depends on it being hang gliding.

For people not familiar with the sports:
- paragliding [https://en.wikipedia.org/wiki/Paragliding](url)
- hang gliding [https://en.wikipedia.org/wiki/Hang_gliding](url)

jimmykane, 2021-04-15:
Hey , I took a look at this issue. I also asked Suunto. 

Unfortunately paragliding is indeed the correct sport name according to the fit SDK for sport number 26 

We can solve this in a different way. I could just edit the string of hang-gliding on quantified self and make it paragliding. 

What do you think ?

jimmykane, 2021-04-15:
See also https://developer.garmin.com/connect-iq/api-docs/Toybox/ActivityRecording.html

NainDeathlegs, 2021-04-16:
But if you replace the string in quantified self, then any task that was set to hang-gliding deliberately will be paragliding now. You would basically eliminate the sport. And if you just replace the two then any entries that were uploaded and just changed to the correct one of the two will be swapped again.
Isn't it easier to fix the parser to make sure that future tasks are correct instead of fixing a symptom?
I'm sure you have access to the full DB - check how many tasks are already there.  I don't think that too many people are as weird as me and even upload paragliding activities here; HR is not really relevant to the sport and all specific platforms use .igc for documentation - which has a different focus

jimmykane, 2021-04-19:
Well this is the debate.

EasyFit (this repo / fork is ahead) should be 1:1 with the Garmin SDK. That means 26 is hang_gliding unfortunately. 

Now Quantified-Self can just change a localization so that hang_gliding -> paragliding. 

That said it's difficult to support both sports since the FIT file protocol does not , and that is the source of the data.

NainDeathlegs, 2021-04-19:
Sorry, I misunderstood that the value comes from the SDK. Seems like Suunto has been using the value for longer than the SDK and is therefore reluctant to update their implementation as there is no alternative that Garmin can't mess with in the future.

Either way there doesn't seem to be a point in changing anything here. Renaming the sport is not an alternative for me personally as I still want to try hang-gliding ;-)

closed 2021-04-19 by NainDeathlegs