jimmykane/fit-parser #28: Undefined function in binary.js located in publish npm package

thomaschampagne, 2021-09-14:
Hey @jimmykane,

I'm looking to understand why every cycling dynamics stats like `avg_left_power_phase`, `avg_left_power_phase_peak` are returning `NaN` values inside this library (file used: [7386755164.zip](https://github.com/jimmykane/fit-parser/files/7163804/7386755164.zip)). 

1 - I noticed  that in the npm distributed package (v1.9.0) the following file `node_modules\fit-file-parser\dist\binary.js` has an error. Indeed inside function `readData` the following line:

```javascript
return dataView.getUnt8(0, fDef.littleEndian);
```
Should be 

```javascript
return dataView.getUint8(0, fDef.littleEndian);
```

`getUnt8` function doesn't exists, the `getUint8` yes !

2 - Aside this, the `readData` from repository is not  the same than the npm package. Why? I was unable to find tag 1.9.0 on this repository.

Thanks for your help !

Thomas

jimmykane, 2021-09-16:
Hey, that's correct it's a bug most probably.

I'll be fixing this somewhere next week if that is ok with you

thomaschampagne, 2021-09-16:
It's perfect :) Thanks !

By the way, I switched from npm package to current repository state `fit-parser` to  debug my cycling dynamics stats issues (avg_left_power_phase=NaN, avg_left_power_phase_peak=NaN, ...). I have the correct code :) But i still have `NaN` issues. 

The activity used to fetch cycling dynamics stats is https://connect.garmin.com/modern/activity/7386755164 (https://github.com/jimmykane/fit-parser/files/7163804/7386755164.zip)

Thanks !

thomaschampagne, 2021-09-17:
You will also find below the CSV export of this fit file using the java `FitCSVTool.jar` provided in latest FIT SDK from garmin

```bash
java -jar FitCSVTool.jar 7386755164.fit
```
Gives: [7386755164_FitCSVTool_export.csv.zip](https://github.com/jimmykane/fit-parser/files/7185134/7386755164_FitCSVTool_export.csv.zip)

Power phase like data is returned as is for instance: `left_power_phase=28.12500043945313|202.50000316406255`. This format could explain the `NaN` results.

Once this npm publish et field fixed I will provide new fields I identified through a PR. And also in a `sports-lib` PR with a lot of improvements I'm preparing for you for many weeks ;)

Thanks again!

Thomas

thomaschampagne, 2021-09-24:
Hey @jimmykane . Have you been able to take a look at this? Thanks to you !

jimmykane, 2021-09-24:
Not yet buddy. Broke my toe and have been a bit strange. Next week should
be fixed. I have also some time free so I promise.

thomaschampagne, 2021-09-24:
Ouch ! Hope you're fine and can walk by the way? It can wait. Have a good recovery !!!

jimmykane, 2021-09-27:
I took a look and strangle this part of issue is not in the code of this repo. :-O

jimmykane, 2021-09-27:
@thomaschampagne  I tested your file after some fixes here and there and works ok. 

I also released sports-alliance/sports-lib v5.4.24 that should have this fixed 

[image]

thomaschampagne, 2021-09-28:
Hi @jimmykane. 

**(1)** About the power I was probably unclear :D I was talking about the L/R power **phase**  which is a angle between 0 to 360 degrees. This is what you can see there (https://connect.garmin.com/modern/activity/7386755164): 

[image]
[image]

This angle/phase L/R data is parsed as `NaN` by the fit-parser (1.9.2):

[image]

I think the issue comes from how the data is formatted inside the FIT protocol (see https://github.com/jimmykane/fit-parser/issues/28#issuecomment-921717808)

**(2)** Just to mention it, I updated to `"fit-file-parser": "^1.9.2"`, and I still have the wrong code (with the `getUnt8` function undefined) in the generated dist folder.

[image]

I hope your toe goes better?

Thanks again,

Thomas

jimmykane, 2021-09-29:
Undefined was fixed. It was the issue in src , strange I had not spotted it. Can you try if this by any way fixes the l_power_phase etc? 

If else I think those fields are not implemented for decoding. This will take a few

jimmykane, 2021-09-29:
@thomaschampagne version 1.9.3 with fix to the unknown function is out

thomaschampagne, 2021-09-29:
> Undefined was fixed. It was the issue in src , strange I had not spotted it. Can you try if this by any way fixes the l_power_phase etc?
> 
> If else I think those fields are not implemented for decoding. This will take a few

I try this this evening! I let you know

thomaschampagne, 2021-09-29:
I updated to 1.9.3. Unknown function is fixed. However it doesn't fix the `NaN` values on power phases

[image]

nenes82, 2022-05-17:
Hello, 
my problem was that the scale value is 0,711111. I replaced comma by a dot and it's ok !
May be there is a better solution to convert value during the execution ?

REgards