jimmykane/fit-parser #31: Support of "summary first" for Garmin upcoming changes

collax, 2023-10-30:
https://forums.garmin.com/developer/fit-sdk/b/news-announcements/posts/important-fit-activity-file-message-change

After the 1s December, Garmin will save the files with the "summary first" method instead of the "summary last". Using their example file, I don't seem to be able to get the data inside the sessions?  Are the files using the summary first method supported?

gie3d, 2023-11-08:
Could you share your problem more specific?

Normally I use this package to get data from Garmin fit file as well (to be more specific, I usually get `data.activity.sessions` and `data.activity.sessions[i].labs` and it seems to be ok with both summary first and summary last.

gie3d, 2023-11-08:
@collax I face the same issue as you. I normally use `cascade` mode when I parse my fit files.

My workaround would be changing to `list` mode. In `list` mode, you won't get the parent-child relationship between sessions, laps and records. But given the timestamp, in sessions, laps and records, I think, you could re-group them again.

krzysztof-cislo, 2023-11-08:
@jimmykane do you think you will be able to fix the issue before December? Or this project is abandoned for you?

jimmykane, 2023-11-08:
It's not abandoned but I have 0 time at the moment due to family issues and
health. Unfortunately

Could you help ?

gie3d, 2023-11-09:
@jimmykane I'll look into this issue. Maybe this or next week. My idea would be changing the logic for `cascade`. 
Because the main different between Summary First and Summary Last is the sequence of message and the timestamp. I think rather than push the data into `temp` and from`temp` into its parent inside the main while loop. I might create a new while loop to create parent child relation.

If you have any suggestion, please let me know :)

I'll try to implement and test this and then submit a PR.

And thank you very much for creating this project :)

krzysztof-cislo, 2023-11-17:
@gie3d did you find some time to look at this issue?

gie3d, 2023-11-17:
> @gie3d did you find some time to look at this issue?

Yes I submitted my pull request. I think @jimmykane might not be able to review it at the moment. In that case, I think you might review from the PR mentioned above if it could solve the problem and create a new package yourself.

krzysztof-cislo, 2023-11-17:
> > @gie3d did you find some time to look at this issue?
> 
> Yes I submitted my pull request. I think @jimmykane might not be able to review it at the moment. In that case, I think you might review from the PR mentioned above if it could solve the problem and create a new package yourself.

Thanks! I will check the PR till EOW. Thanks again!

jimmykane, 2023-11-21:
Guys feel free to close this

krzysztof-cislo, 2023-11-21:
I did some tests and unfortunately after upgrade to 1.9.4 the the output is different...

I used example multisport files from garmin announcment:
https://forums.garmin.com/developer/fit-sdk/b/news-announcements/posts/important-fit-activity-file-message-change

In short when I parse these files (summary first and summary last) with 'both' mode sessions don't contains laps. Laps are in the fit object on the top level, next to sessions but they are not in the tree...

@gie3d can you verify it?

gie3d, 2023-11-21:
@krzysztof-cislo, you're right. There seems to be a problem in my code... will check and fix.

gie3d, 2023-11-22:
@krzysztof-cislo @jimmykane I submitted the PR (#33) also with the json files generated from this and previous version. I'd be great if you guys could double check. I have a very limited number of fit files to test.

krzysztof-cislo, 2023-11-22:
> @krzysztof-cislo @jimmykane I submitted the PR (#33) also with the json files generated from this and previous version. I'd be great if you guys could double check. I have a very limited number of fit files to test.

@gie3d , Why did you add parsed output produced by quite old 1.8.4? The latest version before your changes is 1.9.3

Can you also add one more file to compare? Output of parsed summary last example generated by 1.9.3 version? This could be a reference file to compare as in 1.9.3 summary last fit file is parsed correctly. Then we can compare this PR generated versions (first and last) with 1.9.3 version.

krzysztof-cislo, 2023-11-22:
> @krzysztof-cislo @jimmykane I submitted the PR (#33) also with the json files generated from this and previous version. I'd be great if you guys could double check. I have a very limited number of fit files to test.

@gie3d I checked the PR and the output. Please look at the PR.

jimmykane, 2023-11-23:
1.9.5 is out

krzysztof-cislo, 2023-11-23:
Thank you @gie3d  and @jimmykane !

gie3d, 2023-11-24:
@collax have you tried 1.9.5 if it's ok then you may close this issue.

collax, 2023-11-24:
@gie3d I will test it this weekend, thanks !

collax, 2023-11-25:
Alright it seem to work fine with the example Garmin file. Thanks a lot !

closed 2023-11-25 by collax