jimmykane/fit-parser #34: MessageType 'set' is missing

jastax, 2024-04-29:
Hello,

thanks for this parser. 

The switch (messageType) { } is missing the case 'set' for Garmin Strength workouts, so the records do not show up in the parsed file / laps. 

Could you add it?

gie3d, 2024-08-27:
@jastax hello, I might be able to fix this. Could you share the example fit file for Garmin Strength workouts?

jastax, 2024-08-29:
[1376749965.zip](https://github.com/user-attachments/files/16796692/1376749965.zip)

Here is one example. It's a workout with only one set though. If you need a better one I can create one :)

gie3d, 2024-09-02:
@jastax I looked into the fit file today but still cannot get `set`. Could you elaborate more on the `set` data and would be great if you could visualize it (maybe a screenshot from your fit file app)

I also use the FitCSVTool to export your fit file to .csv (in the attached). I would be great if you could guide me to the `set` data.
[with_set.csv](https://github.com/user-attachments/files/16839504/with_set.csv)