jimmykane/fit-parser #36: Missing field in Garmin Descent file

danvoyce, 2024-06-11:
I'm trying to parse and visualise the `record` data. All works perfect, however I'm missing the `tank pressure` field that has been captured by the [wireless transmitter](https://www.garmin.com/en-GB/p/852159).

The data is definitely there because I can see it visualised in another app.

I\'m not exactly sure of the field name. Any idea how I can expose it? 

Many thanks and awesome library 🙏

gspsteve, 2024-07-02:
Same here, i miss some data after try to parse a garmin fit file. In my case i'm parsing a "strengh_workout" fit file and the parser don't return the "Set" Data.

danvoyce, 2024-08-25:
@gie3d  absolutely. I'm going to email it to you as there may be some sensitive information on it that the owner of the file may not want put on the internet. 

Thanks so much!

jimmykane, 2024-08-26:
Looks like the fit.js does not have the info for the tank pressure.  It could be added though. One needs to look at the FIT sdk for the correct place.

gie3d, 2024-08-26:
@jimmykane I'm working on this. Looking into the fit file and I think this field is located in `tank_update` and inside that field, it also has a `pressure` which is in `bar` unit. I can now extract it but got a few questions. Appreciate if I could get suggestion from you and @danvoyce

From `tank_update`: an array of tank_update, which has `sensor` and `pressure`
```
[{
    "timestamp": "2023-11-22T15:16:06.000Z",
    "sensor": 1111111111,
    "pressure": 100.00
  },
  ...
]
```

From `tank_summary`, we got the `start_pressure` and `end_pressure` for each `sensor`
for example, we have 2 sensors
```
[
        {
            "sensor": 1111111111,
            "start_pressure": 100.00,
            "end_pressure": 50.00
        },
        {
            "sensor": 1234567890,
            "start_pressure": 0,
            "end_pressure": 0
        }
]
```

**The 1st question** is should I combine the `tank_update` into `tank_summary` to get the output like this
```
[
        {
            "sensor": 1111111111,
            "start_pressure": 100.00,
            "end_pressure": 50.00,
            "tank_updates": [{
                        "timestamp": "2023-11-22T15:16:06.000Z",
                        "pressure": 100.00
                        },
                        ... // the rest of tank_updates
            ]
        },
        {
            "sensor": 1234567890,
            "start_pressure": 0,
            "end_pressure": 0,
            "tank_updates": [] // this probably an empty array if there is no record for this sensor
        }
]
```

**The 2nd question** is I set the default unit for pressure to `bar`, are there any units we should support here?

danvoyce, 2024-08-26:
Thanks for picking up @gie3d 🙏

I've personally only used the `list` mode, so I don't have much of an opinion in regards to your first question.

> The 2nd question is I set the default unit for pressure to bar, are there any units we should support here?

Bar makes sense as the default as it's the metric and mostly commonly used unit. The imperial unit is PSI which they use in the USA, so would be a nice addition if possible

gie3d, 2024-08-28:
@jimmykane PR is submitted. Appreciate if you could review https://github.com/jimmykane/fit-parser/pull/37

jimmykane, 2024-08-29:
`1.10.0` should be out @danvoyce  due to @gie3d 's great work

jimmykane, 2024-08-29:
Feel free to close this if things work out well

danvoyce, 2024-08-29:
I'm just in the midst of a big refactor, but will deff have a play with this in the next couple of days!

Thanks so much @gie3d and @jimmykane 🙏

danvoyce, 2024-09-02:
Looks great, thanks again for this!

closed 2024-09-02 by danvoyce